hello-robot/stretch_factory #10: Firmware updater tool fails to flash calibration with protocol mismatch

hello-binit, 2021-07-29:
After flashing the boards to a newer protocol, the `RE1_firmware_updater.py` tool attempts to open the steppers in Stretch Body to use the `read_encoder_calibration_from_YAML()` and `write_encoder_calibration_to_flash()` methods. Because Stretch Body checks for protocol mismatch in the `startup()` method, this part (and any other part where `startup()` is called after flashing in the tool) will fail to continue.

Some ideas:
 * Using a "allow_mismatched_protocol" flag on the `startup(allow_mismatched_protocol=False)` could work, but the startup method also [calls enable_safety, pull_status, and push_command in stepper.py](https://github.com/hello-robot/stretch_body/blob/master/body/stretch_body/stepper.py#L126-L129). Likely, at least pull_status would fail here.
 * Only start up the transport of the stepper. It's all that `write_encoder_calibration_to_flash()` uses to flash the encoder calibration.

aedsinger, 2021-08-02:
The other way to do this is to report an error if the user tries to upgrade/downgrade firmware to a protocol that doesn't match the currently installed Stretch Body. So to do a protocol switch they must:

* Pip install the newer Stretch Body with P1
* Upgrade firmware. Calibration will pass.

This makes it harder to configure a broken system. Thoughts?

binitshah, 2021-08-03:
I like this solution. The `RE1_firmware_updater.py` tool can identify the protocol version of the firmware branch selected by the user (requires parsing [Common.h](https://github.com/hello-robot/stretch_firmware/blob/feature/waypoint_trajectories_py3/arduino/hello_stepper/Common.h#L24)), and then error out if the currently installed version of Stretch Body doesn't support that protocol. Is there an example of a broken system where we'd intentionally want mismatched protocols?

hello-binit, 2021-08-03:
I like this solution. The `RE1_firmware_updater.py` tool can identify the protocol version of the firmware branch selected by the user (requires parsing [Common.h](https://github.com/hello-robot/stretch_firmware/blob/feature/waypoint_trajectories_py3/arduino/hello_stepper/Common.h#L24)), and then error out if the currently installed version of Stretch Body doesn't support that protocol. Is there an example of a broken system where we'd intentionally want mismatched protocols?

aedsinger, 2021-08-03:
Resolved with PR https://github.com/hello-robot/stretch_factory/pull/13

closed 2021-08-24 by hello-binit