hello-robot/stretch_factory #19: Directory nonexistent error with firmware updater tool

hello-binit, 2021-08-24:
Tried downgrading firmware while having P1 Stretch Body installed. 

```
######### Selecting target branch ###########
0:   origin/HEAD -> origin/master
1:   origin/bugfix/remove_sync_timestamps
2:   origin/bugfix/trajectories_active_id
3:   origin/feature/waypoint_trajectories_py3
4:   origin/master

Please enter desired branch id [0]: 0
Selected branch   origin/HEAD -> origin/master

/bin/sh: 1: cannot create origin/master: Directory nonexistent
Checkout out firmware   origin/HEAD -> origin/master from Git for hello_wacc
---------------------------
Target firmware branch of   origin/HEAD -> origin/master is incompatible with installed Stretch Body for device hello-wacc
Installed Stretch Body supports protocol P1
Target branch supports protocol P0
Downgrade Stretch Body first...
hello-binit@stretch-re1-1002:~$
```

The firmware updater tool errors out as expected, but also prints out an `Directory nonexistent` error.

hello-binit, 2021-08-24:
Once Stretch Body is downgraded, we see protocol mismatch errors on running a status check:

```
hello-binit@stretch-re1-1002:~$ RE1_firmware_updater.py --status
[WARNING] [wacc]: 
----------------
Firmware protocol mismatch on hello-wacc.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-lift]: 
----------------
Firmware protocol mismatch on hello-motor-lift.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [pimu]: 
----------------
Firmware protocol mismatch on hello-pimu.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-arm]: 
----------------
Firmware protocol mismatch on hello-motor-arm.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-left-wheel]: 
----------------
Firmware protocol mismatch on hello-motor-left-wheel.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-right-wheel]: 
----------------
Firmware protocol mismatch on hello-motor-right-wheel.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

Already on 'master'
Your branch is up to date with 'origin/master'.
Already up to date.
############## Currently Installed Configuration ##############
------------ HELLO-WACC ------------
Installed Firmware: Wacc.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
------------ HELLO-MOTOR-LIFT ------------
Installed Firmware: Stepper.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
------------ HELLO-PIMU ------------
Installed Firmware: Pimu.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
------------ HELLO-MOTOR-ARM ------------
Installed Firmware: Stepper.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
------------ HELLO-MOTOR-LEFT-WHEEL ------------
Installed Firmware: Stepper.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
------------ HELLO-MOTOR-RIGHT-WHEEL ------------
Installed Firmware: Stepper.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch

######### Currently Tagged Versions of Stretch Firmware on Master Branch ##########
---- HELLO-WACC ----
Wacc.v0.0.1p0
---- HELLO-MOTOR-LIFT ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0
---- HELLO-PIMU ----
Pimu.v0.0.1p0
---- HELLO-MOTOR-ARM ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0
---- HELLO-MOTOR-LEFT-WHEEL ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0
---- HELLO-MOTOR-RIGHT-WHEEL ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0

############## Recommended Firmware Updates ##############
HELLO-WACC                | Downgrade required to Wacc.v0.0.1p0      
HELLO-MOTOR-LIFT          | Downgrade required to Stepper.v0.0.4p0   
HELLO-PIMU                | Downgrade required to Pimu.v0.0.1p0      
HELLO-MOTOR-ARM           | Downgrade required to Stepper.v0.0.4p0   
HELLO-MOTOR-LEFT-WHEEL    | Downgrade required to Stepper.v0.0.4p0   
HELLO-MOTOR-RIGHT-WHEEL   | Downgrade required to Stepper.v0.0.4p0
```

hello-binit, 2021-08-24:
Seeing more `Directory nonexistent` errors:

```
hello-binit@stretch-re1-1002:~$ RE1_firmware_updater.py --update_to_branch
[WARNING] [wacc]: 
----------------
Firmware protocol mismatch on hello-wacc.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-lift]: 
----------------
Firmware protocol mismatch on hello-motor-lift.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [pimu]: 
----------------
Firmware protocol mismatch on hello-pimu.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-arm]: 
----------------
Firmware protocol mismatch on hello-motor-arm.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-left-wheel]: 
----------------
Firmware protocol mismatch on hello-motor-left-wheel.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

[WARNING] [hello-motor-right-wheel]: 
----------------
Firmware protocol mismatch on hello-motor-right-wheel.
Protocol on board is p1.
Valid protocol is: p0.
Disabling device.
Please upgrade the firmware and/or version of Stretch Body.
----------------

Already on 'master'
Your branch is up to date with 'origin/master'.
Already up to date.
############## Currently Installed Configuration ##############
------------ HELLO-WACC ------------
Installed Firmware: Wacc.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
------------ HELLO-MOTOR-LIFT ------------
Installed Firmware: Stepper.v0.0.1p1
Installed Stretch Body requires protocol: p0
Protocol mismatch
[... 583 lines cut ...]
Pimu.v0.0.1p0
---- HELLO-MOTOR-ARM ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0
---- HELLO-MOTOR-LEFT-WHEEL ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0
---- HELLO-MOTOR-RIGHT-WHEEL ----
Stepper.v0.0.1p0
Stepper.v0.0.2p0
Stepper.v0.0.3p0
Stepper.v0.0.4p0

############## Recommended Firmware Updates ##############
HELLO-WACC                | At most recent version with Wacc.v0.0.1p0 
HELLO-MOTOR-LIFT          | At most recent version with Stepper.v0.0.4p0 
HELLO-PIMU                | At most recent version with Pimu.v0.0.1p0 
HELLO-MOTOR-ARM           | At most recent version with Stepper.v0.0.4p0 
HELLO-MOTOR-LEFT-WHEEL    | At most recent version with Stepper.v0.0.4p0 
HELLO-MOTOR-RIGHT-WHEEL   | At most recent version with Stepper.v0.0.4p0 
```

aedsinger, 2021-08-24:
Pushed fix to Develop that removed the offending HEAD branch on --update_to_branch so the 'directory non-existent' errors should be gone. It isn't clear from these logs if there is another issue? They look correct, although their may be a UX issues as they can be confusing to see the 'protocol mismatch' errors.

closed 2021-09-21 by hello-binit