hello-robot/stretch_factory #30: Lift stepper never returns to bus after flash on Python3.8 Ubuntu 20.04

hello-binit, 2021-10-18:
```
-------- FIRMWARE FLASH hello-motor-lift | Stepper.v0.1.0p1 ------------
Starting programming. This will take about 5s...
Success in firmware compile
Success in firmware flash.
Waiting for device hello-motor-lift to return to bus.
.................Failure for device hello-motor-lift to return to USB bus after upload
```

This happens consistently when flashing from Ubuntu 20.04.

aedsinger, 2021-11-22:
Confirmed this issue as well. The following WIP branch is trying to find a solution:

https://github.com/hello-robot/stretch_factory/tree/bugfix/udev_rule

aedsinger, 2022-06-29:
Fixed in 0.2.3. Tested 100 cycles on all boards with no errors (on 20.04)

hello-binit, 2022-09-24:
Seeing the same issue on Ubuntu 20.04, but with additional output from usb.core this time:

```
-------- FIRMWARE FLASH hello-motor-lift | Stepper.v0.2.5p1 ------------
Resetting hello-motor-lift
Looking for device hello-motor-lift on bus
Waiting for device hello-motor-lift to return to bus.
Starting programming. This will take about 5s…
Success in firmware compile
Success in firmware flash.
ls parse: Failed to match all devices for StretchSerialInfo
{‘hello-motor-lift’: {‘device’: None, ‘info’: None, ‘core’: None}}
usb.core parse: Failed to match all devices for StretchSerialInfo
Waiting for device hello-motor-lift to return to bus.
…Failure for device hello-motor-lift to return to USB bus after upload
```

hello-binit, 2022-09-28:
This issue is fixed as of Stretch Factory v0.3.9. Use `pip3 install -U hello-robot-stretch-factory` to update to this version on Ubuntu 20.04

closed 2022-09-28 by hello-binit