hello-robot/stretch_factory #61: REx_robot_trace.py plotting

hello-binit, 2023-04-03:
When attempting to plot arm motor effort for a captured trace, I see a `AttributeError: 'FigureCanvasQTAgg' object has no attribute 'set_window_title'`

Full console log and exception:
```
############### AVAILABLE ################
TRACE    | DURATION(m)               |  START                          |  END                           
--------------------------------------------------------------------------------------------------------------
0        | 0.61                      |  2023-04-03 14:20:22.001372     |  2023-04-03 14:20:58.894312    

ID       | FIELD                     |  SAMPLES                   
--------------------------------------------------------------------------------------------------------------
0        | arm.motor.effort_pct      |  381                       
1        | arm.pos                   |  381                       
2        | arm.vel                   |  381                       
3        | base.left_wheel.effort_pct |  381                       
4        | base.left_wheel.pos       |  381                       
5        | base.left_wheel.vel       |  381                       
6        | base.right_wheel.effort_pct |  381                       
7        | base.right_wheel.pos      |  381                       
8        | base.right_wheel.vel      |  381                       
9        | lift.motor.effort_pct     |  381                       
10       | lift.pos                  |  381                       
11       | lift.vel                  |  381                       
12       | pimu.charger_connected    |  381                       
13       | pimu.cpu_temp             |  381                       
14       | pimu.current              |  381                       
15       | pimu.low_voltage_alert    |  381                       
16       | pimu.runstop_event        |  381                       
17       | pimu.voltage              |  381                       
18       | timestamp                 |  381                       

############### MENU ################
Enter command. (q to quit)
s: set active trace
p: plot
d: print
-------------------------------------
p
############### Plotting ################
ID       | FIELD                     |  SAMPLES                   
--------------------------------------------------------------------------------------------------------------
0        | arm.motor.effort_pct      |  381                       
1        | arm.pos                   |  381                       
2        | arm.vel                   |  381                       
3        | base.left_wheel.effort_pct |  381                       
4        | base.left_wheel.pos       |  381                       
5        | base.left_wheel.vel       |  381                       
6        | base.right_wheel.effort_pct |  381                       
7        | base.right_wheel.pos      |  381                       
8        | base.right_wheel.vel      |  381                       
9        | lift.motor.effort_pct     |  381                       
10       | lift.pos                  |  381                       
11       | lift.vel                  |  381                       
12       | pimu.charger_connected    |  381                       
13       | pimu.cpu_temp             |  381                       
14       | pimu.current              |  381                       
15       | pimu.low_voltage_alert    |  381                       
16       | pimu.runstop_event        |  381                       
17       | pimu.voltage              |  381                       
18       | timestamp                 |  381                       

Enter FIELD ID (range 0 to 18):  0

Traceback (most recent call last):
  File "/home/hello-robot/.local/bin/REx_trace_robot.py", line 189, in <module>
    mgmt.run_menu()
  File "/home/hello-robot/.local/bin/REx_trace_robot.py", line 60, in run_menu
    self.do_plot(trace_data)
  File "/home/hello-robot/.local/bin/REx_trace_robot.py", line 81, in do_plot
    fig.canvas.set_window_title('TRACE')
AttributeError: 'FigureCanvasQTAgg' object has no attribute 'set_window_title'
```

hello-binit, 2023-04-03:
@aedsinger, the Available table is also a little confusing. Currently, it reads:
```
TRACE    | DURATION(m)               |  START                          |  END                           
--------------------------------------------------------------------------------------------------------------
0        | 0.61                      |  2023-04-03 14:20:22.001372     |  2023-04-03 14:20:58.894312
```

I would expect duration to to be the time duration in seconds of the trace. I would call it "Distance traveled(m)". I would replace "END" with "Duration(s)".

hello-binit, 2023-04-03:
Nvm, I confused (m) to be meters. I guess it's actually minutes?

aedsinger, 2023-04-03:
Yes minutes

PR #62 (2023-04-04): Fix trace robot tool plot bug
Fixes #61 and adds better readability to the tool.
closed 2023-04-04 by hello-binit