hello-robot/stretch_factory #95: Firmware Downgrading

hello-binit, 2024-02-25:
`REx_firmware_updater.py` allows advanced users to downgrade their robots firmware. This is useful for testing purposes. 

However, newer versions of the robot will have hardware that requires a certain version of fw, and downgrading past that version could result in unknown behavior or damage to the robot. Therefore, the fw updater should prevent downgrading past a certain version based on the hw ID of the board. The map between ID to min-allowed-version can be hardcoded. Here's the ones I know:

ID #4 - >= v0.7.0p5
ID #3 - ? (Suspect it's v0.7.0p5)
ID #2 - ?
ID #1 - >= v0.3.0p2
ID #0 - ? (Suspect it's v0.0.1p1)

vrathiraj-hellorobot, 2024-02-26:
Hey @hello-binit I think it maybe best to limit the firmware downgrade to v0.3.1p2 reason being is that before this bug fix we were getting usb com issues causing the boards to lock up what do you think?

hello-binit, 2024-02-26:
I wasn't aware that was the case. Yes, I agree, let's use v0.3.1p2 as the lowest version.

closed 2024-03-01 by hello-binit